doebrowsk/tank-chair #1: New Jenks Power Schematic

doebrowsk, 2015-09-25:
Need to copy over / emulate the existing power setup that is present in the ABBY robot. Documentation should be created for this implementation, mainly to detail any specific differences or peculiarities of the task during and after completion.

The schematic from the ABBY thesis should be isolated and given its own file for quick and easy access. It can be found on page 23.
[abby-master-thesis.pdf | uploaded via ZenHub](https://files.zenhub.io/56058df8e7e66e8870a42050)

hdn3, 2015-09-28:
28 Sept 2015 !-- David Notes

Find out how CRIO works. Some cables previously removed and CRIO maybe not be working properly if at all.

Problem with motors draining all power causing CRIO to shut off. Temp solution with capacitor.

Poor wiring throughtout. Look to solve with larger wires and consistent color coding and consistent wire sizes. 

Design with modularity in mind. A lot of components are fixed in place permanently or hard to remove without additionally removing other components.

Power routing is inconsistent. Some cables go through power relay some don't . Some go through CRIO some don't. Maybe the main cause of battery failure within robot.

rkyuan, 2015-09-29:
I hooked up a power supply to where the batteries should be. It is taking 5.5 watts just sitting there which seems rather high.

rkyuan, 2015-10-01:
So we poked at the robot some more today and we basicly understand where all the power is going.
We plugged in the batteries and it seems to be working fine.
The charging also seems to work.
We didn't try having the robot on and charging at the same time.
Here is how the circuit works as we understand it.
[image]

doebrowsk, 2015-10-27:
Please leave a final update on this describing what was accomplished. Better to do while fresher in your mind so that this could be easily drawn upon for the final paper.

closed 2015-12-11 by doebrowsk