doebrowsk/tank-chair #3: Jenks Mast Removal and Replacement

doebrowsk, 2015-09-25:
The masts present on the Jenks frame are too large. Mary (Baxter) will be attached to the top of this robot at some point, so we want to get shorter masts that fit better, and can eventually be cut to size.

Documentation of this effort simply requires saying where the old masts were placed,  and how well the new masts fit.

canoe12, 2015-10-06:
how do I claim this? Cuz I already removed it.

doebrowsk, 2015-10-07:
I've seen that this is finished. Just make a final comment on it, stating where the old mast was stored, and then hit the "Close and comment" button. It should close the issue for us.

canoe12, 2015-10-12:
Mast is completely removed and taken apart and place with the rest of the metal pieces. New frame has been cut to size and installed properly. bottom frame is moved up about an inch for ease of installation and removal in the future

closed 2015-10-12 by canoe12